ros/geometry #84: Rospy: transformBroadcaster.sendTransform cannot be overloaded

lsouchet, 2015-03-23:
Hello,

I recently upgraded my Ubuntu packages (including ros-indigo-geometry), and when I tried to execute a 
transformBroadcaster.sendTransform(position, rotation, time, parent, child), I get this error:
> TypeError: sendTransform() takes exactly 2 arguments (6 given)

When looking at the source files, I found this commit : 
> https://github.com/ros/geometry/commit/924d68822713bf937950c6492a4330c7acc77579

This seems to rely on python inheritance which I don't think exist.

Maybe this commit should be simply reverted.

tfoote, 2015-03-23:
You're right it collides with the other method. It needs to use a different name. probably sendTransformMessage would be better.

tfoote, 2015-03-23:
I'll try to patch this within a day or two. 

FYI @DLu Re #81

commit 52e56e27 (2015-03-25): Broadcaster: Rospy fix #84. Add sendTransformMessage.
commit a1ceb0f5 (2015-03-25): Merge pull request #85 from lsouchet/indigo-devel

Broadcaster: Rospy fix #84. Add sendTransformMessage.
closed 2015-03-25 by tfoote