ros/geometry #93: tf.Exception thrown while using waitForTransform

marcoesposito1988, 2015-04-20:
As reported by other users as well in http://answers.ros.org/question/207039/tfexception-thrown-while-using-waitfortransform a bug seems to have appeared.

waitForTransform throws a tf.Exception without any explanation if used with rospy.Time() or rospy.Time(0) as timeout parameter.

If the time from rospy.Time.now() is used as timeout for waitForTransform and then as time for a lookupTransform, an "extrapolation into the future" exception is thrown.

This bug just popped up today while using code that was perfectly working until the end of March.

marcoesposito1988, 2015-04-21:
It must be a bug in TF2. When I checked out geometry 1.11.4 the bug was still there, but when I also checked out geometry-experimental 0.5.7 the bug disappeared.

Closing here and opening on geometry-experimental.

closed 2015-04-21 by marcoesposito1988

tfoote, 2015-04-21:
Ticket here: https://github.com/ros/geometry_experimental/issues/102

tfoote, 2015-04-21:
Added unit test here: 9624370b2327480ee429c5671455e3880825d256