ros/geometry #96: 'EIGEN3_INCLUDE_DIR-NOTFOUND' not set properly

42f87d89, 2015-05-06:
When building *tf* with catkin_make, the build stops with this error.

    CMake Error at /home/pi/ros_catkin_ws/devel/share/kdl_conversions/cmak /kdl_conversionsConfig.cmake:
    106 (message):
      Project 'kdl_conversions' specifies 'EIGEN3_INCLUDE_DIR-NOTFOUND' as an
      include dir, which is not found.  It does neither exist as an absolute
      directory nor in
      '/home/pi/ros_catkin_ws/src/tf/kdl_conversions/EIGEN3_INCLUDE_DIR-NOTFOUND'.
      Ask the maintainer 'Tully Foote <tfoote@osrfoundation.org>' to fix it.
    Call Stack (most recent call first):
      /home/pi/ros_catkin_ws/devel/share/catkin/cmake/catkinConfig.cmake:75 (find_package)
      tf/tf_conversions/CMakeLists.txt:5 (find_package)

    CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
    Please set them or make sure they are set and tested correctly in the CMake files:
    EIGEN3_INCLUDE_DIR
       used as include directory in directory /home/pi/ros_catkin_ws/src/tf/eigen_conversions
       used as include directory in directory /home/pi/ros_catkin_ws/src/tf/kdl_conversions

    make: *** [cmake_check_build_system] Error 1
    Invoking "make cmake_check_build_system" failed

tfoote, 2015-05-06:
Can you provide instructions how to reproduce this error? kdl_conversions has no references to eigen3 in the package.xml or CMakeLists.txt

42f87d89, 2015-05-07:
I'm not sure.

While trying to build gmapping, I kept adding any package that was in the dependency tree manually, and attempting to build after each addition.

Once I got to tf, I've been stumped by this error.

42f87d89, 2015-05-07:
Ok, if I remove the gmapping package from my catkin src directory, I can build it. But once I add the gmapping package this error occurs.

Why in the world does this error mention tf but not gmapping?

tfoote, 2015-05-07:
At the bottom of the stack trace it's actually a problem when evaluating the tf_conversions which has a dependency on eigen3 which you must be missing. 

Instead of randomly installing dependencies I suggest you checkout the [rosinstall generator](http://wiki.ros.org/rosinstall_generator)

42f87d89, 2015-05-08:
Thank you for your time.

I apologise for opening an issue that was not related to your project.

tfoote, 2015-05-08:
no problem. The traceback was not clear.

closed 2015-05-08 by tfoote