ros/geometry #97: Quaternion getAngle() bugged

AlexReimann, 2015-05-11:
Sorry for having no example code, but I still want to drop this here.

I ran into a bug with getAngle() when continuously getting the robot yaw from tf.

Basically the following code snippet (untested example code) printed two different results:

    tf::Stamped<tf::Pose> global_robot_pose;
    ros::Time time = ros::Time::now();
    tf_listener_->waitForTransform("map", "base_footprint", time, ros::Duration(0.05));
    tf_listener_->transformPose("map", time, costmap_robot_pose, "base_footprint", global_robot_pose);
    
    ROS_INFO("getAngle: %f", global_robot_pose.getRotation().getAngle());
    ROS_INFO("getYaw: %f", tf::getYaw(global_robot_pose.getRotation());

This happens when the angle goes lower than 0. getAngle() didn't give a negative angle, but just went positive again. The result of getRotation works correctly.

tfoote, 2015-05-11:
This is not a bug but expected behavior. getAngle returns the "angle of rotation represented by this quaternion" which is effectively the magnitude of the angle. It does not define direction. Note if you rotated upward it would return you the magnitude of the angular difference yet the yaw would be zero.  

If you want the heading angle about the z axis tf::getYaw is the right method.

closed 2015-05-11 by tfoote

AlexReimann, 2015-05-11:
Ok, thanks!