ros/geometry #98: TF Publisher in Hydro is Blocking

kshehata, 2015-05-11:
Ran into this problem while monitoring a large robot using RViz over a WiFi connection that went flakey. Apparently, TF broadcasting in Python / Hydro is still blocking. This means that when the wifi connection got flakey, the Applanix driver (written in Python) blocked, and our controller stopped being able to do any updates.

From what I can tell in https://github.com/ros/ros_comm/issues/169, ros_comm has been updated, and it looks like there are commits in indigo-devel to fix this. Really it's just a matter of adding queue size to the TF publishers, as per the two commits below. Any chance we can get this back ported to hydro? Seems like a problem many other people are going to face, and it's certainly a major problem on this project.

https://github.com/ros/geometry/commit/90a0fc41380a81b0159d7cb1c1a127088e522543
https://github.com/ros/geometry/commit/fd04fd95a2bf7612aa125a1d40feb37d3fee63e8

tfoote, 2015-05-11:
Yeah, we could backport the queue size. If you have a moment to open a pull request for that it would be great.

tfoote, 2015-05-12:
Fixed in #99

PR #99 (2015-05-12): TF queue size
Changing queue size in TF Publishers to 1 so that publishing is asynchronous. Resolves #98
closed 2015-05-12 by tfoote