ros/geometry #102: Quaternion::getAngleShortestPath

ex-ratt, 2015-07-09:
This function seems to deliver wrong values. According to the description, it should return the angle of rotation along the shortest path - so the magnitude of the angle should never be greater than &pi;. In reality, sometimes the values are even 2&pi;, as you can see with a little example `ROS_INFO("%.3f", tf::Quaternion::getIdentity().getAngleShortestPath());`, which yields `[ INFO] [1436444520.616984435]: 6.283`.

This function is implemented like this:

    tfScalar s;
    if (dot(*this) < 0)
        s = tfScalar(2.) * tfAcos(m_floats[3]);
    else
        s = tfScalar(2.) * tfAcos(-m_floats[3]);
    return s;

On the one hand I think the dot-product with itself cannot get negative, on the other hand the condition should be about `m_floats[3]` (and the negation should be the other way around):

    tfScalar s;
    if (m_floats[3] < 0)
        s = tfScalar(2.) * tfAcos(-m_floats[3]);
    else
        s = tfScalar(2.) * tfAcos(m_floats[3]);
    return s;

In this case, the angles seem to be reasonable. A short example:

    tf::Quaternion q(tf::Vector3(0.5, 0.5, 0.5), tfScalar(0.5));
    double angle = q.getAngleShortestPath();
    double angle2 = q.w() < 0 ? 2 * tfAcos(-q.w()) : 2 * tfAcos(q.w());
    ROS_INFO("q=(%.3f, %.3f, %.3f, %.3f), angle=%.3f/%.3f", q.w(), q.x(), q.y(), q.z(), angle, angle2);

With output:

    [ INFO] [1436445597.807808307]: q=(0.969, 0.143, 0.143, 0.143), angle=5.783/0.500

For comparison, you can have a look at the implementation of this function inside Commons Math https://git1-us-west.apache.org/repos/asf?p=commons-math.git;a=blob;f=src/main/java/org/apache/commons/math4/geometry/euclidean/threed/Rotation.java;h=1a7b0aba76fb3a0dae7b706c9bd86f2a21935205;hb=HEAD#l504

wjwwood, 2015-07-14:
Thanks for the report @ex-ratt. Do you think this issue is related to:

https://github.com/ros/geometry/pull/6

@tfoote is traveling and I'm sure he'll have a look at this as soon as he's able.

ex-ratt, 2015-07-14:
I think your change to getAngleShortestPath would resolve this issue. The major difference is the use of `dot(getIdentity())` instead of just `m_floats[3]`, but as the identity Quaternion is (0, 0, 0, 1), it should be equivalent. I'm just wondering whether the dot-product will make the code more readable or comprehensible - in that case you could also just call `angleShortestPath(getIdentity())` instead and prevent duplicated code (while also doing even more unnecessary computations).

MichaelKorn, 2016-02-15:
I just saw this report. Yesterday, I proposed a patch for this issue: #116 
But I think the proposed changes here have a sightly better performance .

MichaelKorn, 2016-02-15:
I just amended my pull request. The old proposed code is:
`
		tfScalar s = tfScalar(2.) * tfAcos(m_floats[3]);
		if (s > M_PI)
			s = tfFabs(s - 2*M_PI);
`
But the new commit needs less operations.

commit 1577a159 (2016-03-07): fix for issue in getAngleShortestPath(), closes #102

Testing squared norm for <0 does not make any sense. In Addition giving the inverval of the return values is a great help to understand the descriptions.
closed 2016-03-07 by tfoote