ros/geometry #103: TransformerROS.setTransform fails silently with invalid quaternion.

tappan-at-git, 2015-07-31:
I ran into this trying to test some basic tf stuff without thinking too much about creating valid quaternions. tf.TransformerROS.setTransform will fail silently when passed a transform object with an invalid rotation field. The failure is reasonable, but it should raise an exception. Trying to use one of the nonexistent frames later will throw a LookupException, since those frames do not exist, but this does not help track down the root issue.

'''
import rospy
import tf
from geometry_msgs.msg import *
rospy.init_node("foo")
tr = tf.TransformerROS()
p = TransformStamped()
p.header.frame_id = 'world'
p.child_frame_id = 'p'
p.transform.rotation.w = 1    # a basic, valid quaternion (0,0,0,1)
tr.setTransform(p)
print tr.getFrameStrings()    # prints ['q', 'world']

q = TransformStamped()
q.header.frame_id = 'world'
q.child_frame_id = 'q'
# q has rotation = (0,0,0,0), _not_ a valid quaternion, this will not create the q frame but fails silently
tr.setTransform(q)
print tr.getFrameStrings()    # still prints ['q', 'world']
'''

roehling, 2018-07-12:
This issue is kind of resolved since tf2::BufferCore logs a prominent warning `TF_DENORMALIZED_QUATERNION` to the rosconsole. Maybe close the issue?

tfoote, 2018-07-12:
Thanks yes now that the implementation is switched under the hood that should resolve this.

closed 2018-07-12 by tfoote