ros/geometry #108: Is arguments of lookupTransform opposite?

garaemon, 2015-10-22:
tf document and header says:
```c++
void Transformer::lookupTransform(const std::string& target_frame, 
                                           const std::string& source_frame,
                                           const ros::Time& time, StampedTransform& transform) const
```

But in tf_echo, the order looks opposite.
https://github.com/ros/geometry/blob/indigo-devel/tf/src/tf_echo.cpp#L107
```c++
echoListener.tf.lookupTransform(source_frameid, target_frameid, ros::Time(), echo_transform);
```

cc: @wkentaro

wkentaro, 2015-10-22:
I have tested the `lookupTransform` with rviz and `static_transform_publisher` below, and I think these are target/source_frame are opposite.
https://github.com/jsk-ros-pkg/jsk_recognition/pull/1274

[image]

tfoote, 2015-10-22:
It depends on if your referring to the transform of the coordinate or a point between the coordinate frames. 

`tf_echo` and the broadcasters are focused on the transforms between the coordinate frames. When you're building the tree that is what you want to pay attention to. 

When you're using the tf tree you care about transforming data. `lookupTransform` is looking up the transform for an object in the source frame, into the target frame. Which is the inverse of the transform of the coordinate frames that tf_echo would report or that you would publish.

closed 2015-10-22 by tfoote