ros/geometry #109: `euler_from_quaternion` doesn't play nicely with `geometry_msgs.msg.Quaternion`

Jorge-C, 2015-11-02:
I'm happy to try and write a patch if this is unexpected.

```python
In [1]: from tf.transformations import euler_from_quaternion
In [2]: from geometry_msgs.msg import Quaternion
In [3]: euler_from_quaternion(Quaternion(0, 0, 0, 1))
---------------------------------------------------------------------------
TypeError                                 Traceback (most recent call last)
<ipython-input-3-0e8833944c57> in <module>()
----> 1 euler_from_quaternion(Quaternion(0, 0, 0, 1))

/opt/ros/indigo/lib/python2.7/dist-packages/tf/transformations.pyc in euler_from_quaternion(quaternion, axes)
   1095 
   1096     """
-> 1097     return euler_from_matrix(quaternion_matrix(quaternion), axes)
   1098 
   1099 

/opt/ros/indigo/lib/python2.7/dist-packages/tf/transformations.pyc in quaternion_matrix(quaternion)
   1180 
   1181     """
-> 1182     q = numpy.array(quaternion[:4], dtype=numpy.float64, copy=True)
   1183     nq = numpy.dot(q, q)
   1184     if nq < _EPS:

TypeError: 'Quaternion' object has no attribute '__getitem__'
```

tfoote, 2015-12-08:
This may be a documentation issue I believe it's designed on operate on the tf.transformations Quaternion type not the geometry_msgs type. It might be possible to extend it.

chriswl, 2015-12-09:
#100 seems related.

astromme, 2017-11-15:
For other people who are simply trying to apply the function, this works:

```
explicit_quat = [quaternion.x, quaternion.y, quaternion.z, quaternion.w]
your_euler = tf.transformations.euler_from_quaternion(explicit_quat)
```

tfoote, 2017-11-15:
An explicit conversion is necessary. Since there's a simple one line workaround I'm going to close this as won't fix.

closed 2017-11-15 by tfoote

OAkyildiz, 2021-07-30:
except not extending this results in ugly, unpythonic code that adds several lines especially when you are also translating backwards