ros/geometry #111: tf_echo argument inconsistency

gaya-, 2015-11-30:
In ``tf_echo`` the source frame and target frame are mixed up, namely, the parent frame is called the "source frame" and the child is the "target", whereas everywhere else in TF and TF2 the convention is the opposite.

E.g. [here](https://github.com/ros/geometry/blob/indigo-devel/tf/src/tf_echo.cpp#L107) the first argument is called "source_frameid" but it actually is the target frame id.

This inconsistency is really confusing, especially as it's not just the ``tf_echo`` but also the TF tutorials use the convention opposite from the actual implementation.

Should I make a PR for ``tf_echo`` and fix the tutorials afterwards?

tfoote, 2015-11-30:
Duplicate: #108

closed 2015-11-30 by tfoote

gaya-, 2015-12-01:
My bad, only checked the list of open issues.

Using "parent_frame" and "child_frame" instead of "source" and "target" might help to alleviate the confusion in the future...