ros/geometry #117: The tf python api is not compatible with tf2's /tf_static

sloretz, 2016-02-26:
The python tf api on Jade cannot lookup transforms published on /tf_static.
Section 1.8 on the [tf2 migration page](http://wiki.ros.org/tf2/Migration) says 

> the tf implementation has been removed and replaced with calls to tf2 under the hood. This will mean that all users will be compatible with tf2

I would expect that to mean the tf api would be aware of /tf_static transforms.

How to reproduce:

1. Start roscore and publish a static transform

 `rosrun tf2_ros static_transform_publisher 0 0 0 0 0 0 parent_frame child_frame`
2. Start python and run the following code using the tf api

 ```
 import tf
 import rospy
 rospy.init_node("quick_test")
 listener = tf.TransformListener()
 listener.lookupTransform("parent_frame", "child_frame", rospy.Time())
 # Traceback (most recent call last):
 #  File "<stdin>", line 1, in <module>
 # tf.LookupException: "parent_frame" passed to lookupTransform argument target_frame does not exist. 
 ```

3. Start another instance of python and run the equivalent code on the tf2 api

 ```
 import tf2_ros
 import rospy
 rospy.init_node("quick_test2")
 tfBuffer = tf2_ros.Buffer()
 listener = tf2_ros.TransformListener(tfBuffer)
 tfBuffer.lookup_transform("parent_frame", "child_frame", rospy.Time())
 # header: 
 #   seq: 0
 #   stamp: 
 #     secs: 0
 #     nsecs: 0
 #   frame_id: parent_frame
 # child_frame_id: child_frame
 # transform: 
 #   translation: 
 #     x: 0.0
 #     y: 0.0
 #     z: 0.0
 #   rotation: 
 #     x: 0.0
 #     y: 0.0
 #     z: 0.0
 #     w: 1.0
 ```

lucasw, 2016-05-19:
This also manifests as a `tf.ConnectivityException: Could not find a connection between 'link1' and 'link2' because they are not part of the same tree.Tf has two or more unconnected trees.` exception for tf trees dynamic links trees on either side of a static.

sloretz, 2016-09-15:
I tested it again today. It's still an issue on Kinetic
```
$ apt show ros-kinetic-tf
Package: ros-kinetic-tf 
Version: 1.11.8-0xenial-20160721-134129-0700  
```

jonbinney, 2016-10-28:
Just ran into this  - @tfoote should static frames work in python?

tfoote, 2016-10-28:
Looking at this I think that the tf python bindings have not been extended to subscribe to the static transform topics. I'd suggest using tf2_ros' python bindings instead: http://wiki.ros.org/tf2/Tutorials/Writing%20a%20tf2%20listener%20%28Python%29

jonbinney, 2016-10-28:
Can the python library for tf be officially deprecated then? I think a lot of people still use it, and won't realize that it doesn't work with static transforms anymore (specifically since https://github.com/ros/robot_state_publisher/pull/56)

sloretz, 2016-10-30:
@tfoote What do you think about replacing the python tf implementation with calls to tf2_ros?

tfoote, 2016-10-30:
That would be the optimal solution then they would stay in sync.

roehling, 2016-11-24:
Since we have been battling with the same problem for quite some time, I'd be willing to provide a pull request for that.

rconnorlawson, 2017-06-15:
I just found this bug under ROS Kinetic on Ubuntu 16.04. Looks like merge #134 has had the solution ready since early March. Can this be merged and closed?

tfoote, 2017-07-14:
Resolved with #134 being merged

closed 2017-07-14 by tfoote