ros/geometry #120: tf2::lookupTransform with time 0 throws Extrapolation

cyril42e, 2016-03-11:
Here is my code, I first try to lookup the transform at some date, and if there is an extrapolation exception (usually because I'm trying to get in the future) I just ask for the latest tranform :

		tf2_ros::Buffer m_tf_buffer;
		tf2_ros::TransformListener m_tf_listener(m_tf_buffer);

		std::cout << "BEGIN" << std::endl;
		geometry_msgs::TransformStamped tf_pose;
		try
		{
				try
				{
						std::cerr << "getTranform at " << in_stamp.toSec() << std::endl;
						tf_pose = m_tf_buffer.lookupTransform(frame_id1, frame_id2, ros::Time(date));
				}
				catch (const tf2::ExtrapolationException& e)
				{
						std::cerr << e.what() << std::endl;
						std::cerr << "getTranform at *0* << std::endl;
						tf_pose = m_tf_buffer.lookupTransform(frame_id1, frame_id2, ros::Time(0.0));
				}
		}
		catch (const tf2::ExtrapolationException& e)
		{
				std::cerr << "ERROR should not throw tf2::ExtrapolationException here ! : " << e.what() << std::endl;
				tf_pose = m_tf_buffer.lookupTransform(frame_id1, frame_id2, ros::Time(0.0));
				exit(1);
		}

The problem is that when I ask for the latest transform, I sometimes get an extrapolation exception because the "asked" date is too old (even though I did not ask any date, only the latest).

The output of the previous code is :

>BEGIN
>getTranform at 13.37
>Lookup would require extrapolation into the future.  Requested time 13.370000000 but the latest data is at time 13.320000000, when looking up transform from frame [vehicle] to frame [world]
getTranform at *0*
>ERROR should not throw tf2::ExtrapolationException here ! : Lookup would require extrapolation into the past.  Requested time 13.320000000 but the earliest data is at time 13.334000000, when looking up transform from frame [vehicle] to frame [world]
terminate called after throwing an instance of 'tf2::ExtrapolationException'
>  what():  Lookup would require extrapolation into the past.  Requested time 13.320000000 but the earliest data is at time 13.334000000, when looking up transform from frame [vehicle] to frame [world]

We can notice that the first exception states that the latest data is at time 13.32, but then the second exception states that the earliest data is at time 13.334, which is contradictory...

tfoote, 2016-03-11:
A time query of zero asks for the latest time at which the spanning set can be computed. However if there is data for the whole span but no common time then I believe you can get this exception.

You are not giving the buffer any significant time to collect data. As such I suspect that you have approximately 1 datapoint per frame_id when querying and there's no common time. Thus when you query for the latest common time, it finds a time which is older than one of the links has data and thus it throws an exception.

cyril42e, 2016-03-11:
I see what you mean, makes sense actually! If I wait that it succeeded once to rethrow the exception it does not seem to happen later. Thank you for your help, my apologies for this invalid issue (should have asked the question on the forum before).

closed 2016-03-11 by cyril42e

thecoldviews, 2019-07-05:
Can you elaborate on what solution you came up with @cyril42e . I think I am having the same issue, but I didn't get how you solved it?