ros/geometry #136: tf_monitor and view_frames missing tf_static transformations generated using robot_state_publisher

piyushk, 2016-12-10:
ROS Kinetic, tf v1.11.8 (Xenial)

I did an update recently, and some frames have stopped appearing in `view_frames` and `tf_monitor`, whereas `tf_echo` and rviz visualization seems to work correctly. As far as I can tell, this has something to do with tf frames being published in tf_static. I'm sharing a reproducible example here, where no frames are seen in `view_frames` and `tf_monitor`. In my actual use-case, some frames appear and some don't.

```
roscore
rosparam load segbot_v2.urdf robot_description
rosrun robot_state_publisher robot_state_publisher
```
[segbot_v2.urdf.txt](https://github.com/ros/geometry/files/643958/segbot_v2.urdf.txt)

Here's the relevant output:

```
rosrun tf view_frames && evince frames.pdf
```
[frames.pdf](https://github.com/ros/geometry/files/643959/frames.pdf)

```
piyushk@roberto:~/catkin_ws/src/segbot/segbot_bringup$ rosrun tf tf_monitor

RESULTS: for all Frames

Frames:

All Broadcasters:

RESULTS: for all Frames

Frames:

All Broadcasters:
```

Correct output using `tf_echo`:
```
piyushk@roberto:~/catkin_ws/src/segbot/segbot_bringup$ rosrun tf tf_echo base_footprint base_link
At time 0.000
- Translation: [0.000, 0.000, 0.216]
- Rotation: in Quaternion [0.000, 0.000, 0.000, 1.000]
            in RPY (radian) [0.000, -0.000, 0.000]
            in RPY (degree) [0.000, -0.000, 0.000]
```

The problem corrects itself if I set the `use_tf_static` parameter in `robot_state_publisher` to false.

gavanderhoorn, 2017-01-02:
I think [No transform between base_footprint and base_link](http://answers.ros.org/question/249766) on ROS Answers describes the same problem.

clydemcqueen, 2017-01-02:
I was able to reproduce this. I can also reproduce this with a simple urdf file with 2 links and 1 fixed joint between them.

I think it's related to this: https://github.com/ros/geometry/issues/117
 
Kinetic, Xenial
tf 1.11.8
tf2 0.5.13
robot_state_publisher 1.13.3

commit 868fd89c (2017-07-14): Add support for static_transforms in tf_monitor.

Fixes #136 with #134 for tf_echo and view_frames.
closed 2017-07-14 by tfoote