ros/geometry #139: eigen_conversions does not have a method for msg to Eigen matrix

anuppari, 2017-02-08:
There is a function for converting a generic Eigen matrix to a `std_msgs` multidimensional array [here](https://github.com/ros/geometry/blob/4861b35f4c556def1f924d6d7e079b3c327b5586/eigen_conversions/include/eigen_conversions/eigen_msg.h#L107), but no corresponding function to go back the other way.

tfoote, 2017-02-08:
I could consider merging a PR to add that. However both this package is deprecated for http://wiki.ros.org/tf2_eigen and the std_msgs MultiDimentionalArrays datatypes are generally not recommended either.

anuppari, 2017-02-08:
If that message type is not recommended, and since tf2 focuses on geometric types, what's the best way to send generic Eigen matrices over ROS topics?

tfoote, 2017-02-13:
We generally discourage sending generic matrices as messages since they are not semantically meaningful. The MultiDimentialArrays are likely the best way to do that, but you need to keep in mind that the messages might be logged or otherwise used and are best being fully self contained and documented.

closed 2017-07-14 by tfoote