ros/geometry #141: `tf.transformations.rotation_from_matrix` - Out of index bug, potentially redundant code

eacousineau, 2017-03-07:
It appears that there may be an out of index bug with `rotation_from_matrix`:

```
>> import tf.transformations as tform
>> import numpy.matlib as nm
>> R = nm.eye(3)
>> angle, direction, point = tform.rotation_from_matrix(R)
Traceback (most recent call last):
  File "test.py", line 9, in <module>
    angle, direction, point = tform.rotation_from_matrix(R)
  File "/opt/ros/indigo/lib/python2.7/dist-packages/tf/transformations.py", line 346, in rotation_from_matrix
    point /= point[3]
IndexError: index out of bounds
```

The fix is to change the offending line to:
```
point /= point[-1]
```

That being said, it isn't really clear as to:

1. What is special about `direction` versus `point` when `R` itself is  a 3x3 rotation matrix. It also is not clear as to why a larger matrix is permitted.
2. Why it is dividing by the last element. I'm a little rusty on linear algebra, but I believe this should be normalized with `numpy.linalg.norm`, but even that seems redundant.

Tried `git blame`, but it appears that this code hunk hasn't been touched since it's first relevant commit back in 2009.

Should this code be trimmed?

eacousineau, 2017-03-07:
Saw the scope of the library. Closing the issue.

closed 2017-03-07 by eacousineau

tfoote, 2017-03-07:
This code is imported from: http://www.lfd.uci.edu/~gohlke/code/transformations.py.html

I'd recommend checking out transforms3d: https://github.com/matthew-brett/transforms3d and https://pypi.python.org/pypi/transforms3d as an updated fork.

tfoote, 2017-03-07:
Also there's been some work toward packaging it for ROS too: https://github.com/ros/rosdistro/pull/12437