ros/geometry #144: fatal error: XmlRpcValue.h: No such file or directory compilation terminated.

vannem95, 2017-04-29:
Catkin_make is giving this error:

**catkin_ws/src/geometry/tf/src/change_notifier.cpp:34:25: fatal error: XmlRpcValue.h: No such file or directory
compilation terminated.
geometry/tf/CMakeFiles/tf_change_notifier.dir/build.make:62: recipe for target 'geometry/tf/CMakeFiles/tf_change_notifier.dir/src/change_notifier.cpp.o' failed
make[2]: *** [geometry/tf/CMakeFiles/tf_change_notifier.dir/src/change_notifier.cpp.o] Error 1
CMakeFiles/Makefile2:45425: recipe for target 'geometry/tf/CMakeFiles/tf_change_notifier.dir/all' failed
make[1]: *** [geometry/tf/CMakeFiles/tf_change_notifier.dir/all] Error 2
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j8 -l8" failed**

I can't tell what the issue is. Anyone got any ideas?

tfoote, 2017-05-01:
The error suggests that you're missing a build dependency, specifically `xmlrpcpp`. With continuous integration and binary builds passing I don't think we have a bug at this level. Please as for help on answers.ros.org and in your question provide enough information to reproduce your problem with a minimal example. This should include at minimum your platform, your installation, and your workspace setup as wall as the command you actually ran and the full contents of the output not just the error.  

I'm going to close this ticket. If there is found to be a bug please comment back here and we can reopen this issue.

closed 2017-05-01 by tfoote

gerkey, 2017-05-01:
There's a chance that this problem is caused by ros/ros_comm#962, which relocated the xmlrpcpp headers. I believe that all the necessary consequent fixes have been made and released, but you might need to update some things to get those fixes.

tfoote, 2017-05-01:
There's a similar answers.ros.org question here too: http://answers.ros.org/question/260022/which-library-can-i-access-xmlrpcvalueh-file/

mightyCelu, 2017-05-03:
I, too, believe this is caused by changes in ros-comm as this commit showcases: https://github.com/ros/ros_comm/pull/962/commits/93b64b854c50f148244b0354a85cceece18f3752. The old locations are exported for compatibility, but I believe one should adapt to the changes as soon as possible.

I prepared the (trivial) change in a PR.

cyberguy42, 2017-10-10:
Unless I'm doing something else wrong, this problem still exists when compiling ROS from source on Ubuntu 16.04. I downloaded the packages today using these commands:

```
rosinstall_generator desktop_full turtlebot_gazebo --rosdistro kinetic --deps --wet-only > kinetic.rosinstall
wstool init -j500 src kinetic.rosinstall
```

Other relevant information:
`tf` is on branch `release/kinetic/tf/1.11.9-0`.
`ros_comm/xmlrpcpp` is on branch `release/kinetic/xmlrpcpp/1.12.7-0`.

The diagnostic_aggregator package has a similar issue (branch `release/kinetic/diagnostic_aggregator/1.9.2-0`) at these locations:
```
diagnostic_aggregator/src/analyzer_group.cpp:37:0
diagnostic_aggregator/include/diagnostic_aggregator/generic_analyzer.h:55:25
```

Adding `include/xmlrpcpp` to the `catkin_package INCLUDE_DIRS` in `ros_com/xmlrpcpp/CMakeLists.txt:12` as suggested [here](https://github.com/ros/ros_comm/pull/962#issuecomment-273510911) resolves the issue for me.

Let me know if there is any other information that would be helpful.