ros/geometry #148: tf SyntaxError: Missing parentheses in call to 'print'

Alessandro-Barbieri, 2017-06-24:
when building from source

```
running install_scripts
copying /home/ale/catkin_ws/build_isolated/tf/scripts-3.5/tf_remap -> /opt/ros/lunar/bin
copying /home/ale/catkin_ws/build_isolated/tf/scripts-3.5/view_frames -> /opt/ros/lunar/bin
changing mode of /opt/ros/lunar/bin/tf_remap to 755
changing mode of /opt/ros/lunar/bin/view_frames to 755
running install_egg_info
Writing /opt/ros/lunar/lib64/python3.5/site-packages/tf-1.11.8-py3.5.egg-info
  File "/opt/ros/lunar/lib64/python3.5/site-packages/tf/tfwtf.py", line 181
    print "running tf checks, this will take a second..."
                                                        ^
SyntaxError: Missing parentheses in call to 'print'
```

tfoote, 2017-06-26:
This looks like you're using python3.5. It's something that we try for compatibility but is not the default platform. If you can submit a PR to make it 2/3 compatible that would be appreciated.

Alessandro-Barbieri, 2017-06-26:
Sorry my default python was 3.5, after finding that ROS needs python2 I have switched my default python to 2.7 and now works.
Feel free to close this issue.

tfoote, 2017-07-14:
Closing as duplicate of #67

closed 2017-07-14 by tfoote