ros/geometry #154: Error in getFrameStrings with release 1.11.9

davidmball, 2017-08-18:
We updated to 1.11.9 and in this code
```
                _print("Waiting for %s frame..." % to_frame)
                while not transform_listener.frameExists(to_frame):
                    if rospy.is_shutdown():
                        return
```
now have the following error
```
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/tf/listener.py", line 93, in frameExists
    return frame_id in self.getFrameStrings()
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/tf/listener.py", line 98, in getFrameStrings
    return [p for p, _ in data.items()]
AttributeError: 'list' object has no attribute 'items'
```
Switching to ```canTransform``` solved our problem.

tfoote, 2017-09-27:
Can you confirm what the line 97 reads in listener.py? That issue should have been resolved by #153 

The only case that the data coming out of that function should be a list is the empty list. All other elements provide a dict. https://github.com/ros/geometry2/blob/indigo-devel/tf2/src/buffer_core.cpp#L1195

tfoote, 2018-02-14:
Resolved 2 ways, the upstream way is released, and in the next release here it will be more defensive too.

closed 2018-02-14 by tfoote