ros/geometry #155: Ubuntu 16.04 terminal messed up after running static_transform_publisher

vik748, 2017-10-20:
I have noticed on multiple machines running Ubuntu 16.04 that when running static_transform_publisher from command line and ctrl-C'ing out of it, the terminal becomes messed up.  What I mean is that when you press the up arrow key and change a parameter say the roll angle and hit enter, it actually runs values different from what is intended.  These can then be observed by ctrl-C'ing out again and pressing the up arrow.
Basically the terminal doesn't execute what it displays.  Has anyone else observed this behavior?

tfoote, 2017-10-20:
I've seen this behavior when using many long command line invocations that wrap. Sometimes the terminal doens't clear/rewrap correctly. I suspect you'll find that it's shifted by a couple characters due to a different prompt length or other justification issues. You can clear the issue by forcing the terminal to rerender, or making it wide enough that it doesn't wrap.

tfoote, 2017-11-15:
I don't believe that this is an issue with tf, but with the terminal.

closed 2017-11-15 by tfoote