ros/geometry #161: Where's the official Python API documentation for TF?

nbro, 2018-04-07:
Please, see the following question: https://answers.ros.org/question/287826/wheres-the-official-python-api-documentation-for-tf/.

In a nutshell, I think it's quite difficult to find the documentation for the Python API for TF, the documentation is poorly written or quite incomplete. I think this is a big issue which should be solved. Also, you should have the documentation for each distribution of ROS.

tfoote, 2018-04-10:
All the documentation from that page can be found under the standard "Code API" link which is uniform for all documented packages. And there are versions of that documentation for all rosdistros can be reached within 3 clicks on auto generated standard links as pointed out by 

This is an open source project if you'd like to make a contribution that is appreciated. Asserting things are not good enough for you is not productive or helpful, especially if you're assertions are incorrect.

Please read our [support guidelines](http://wiki.ros.org/Support) for how to ask questions. I have noticed at least [one other](https://answers.ros.org/question/287855/whats-interpolation-and-extrapolation-in-the-context-of-ros/) instance where your post are combative and inflammatory, these posts are not appropriate and do not follow our guidelines.

closed 2018-04-10 by tfoote

nbro, 2018-04-10:
> All the documentation from that page can be found under the standard "Code API" link which is uniform for all documented packages. And there are versions of that documentation for all rosdistros can be reached within 3 clicks on auto generated standard links as pointed out by

> This is an open source project if you'd like to make a contribution that is appreciated. Asserting things are not good enough for you is not productive or helpful, especially if you're assertions are incorrect.

@tfoote Explain to me the upvote of my question and the reason for having multiple answers, then we talk about visibility and incorrectness of my statements.

It's open source, but it is led by people like you and who own the repositories. 

> I have noticed at least one other instance where your post are combative and inflammatory

I am not going to accept an answer which is not answering to my question. Forget about it!

tfoote, 2018-04-10:
> I am not going to accept an answer which is not answering to my question. Forget about it!

No one has asked you to accept an answer that does not answer your question. However in [that thread](https://answers.ros.org/question/287855/whats-interpolation-and-extrapolation-in-the-context-of-ros/?answer=287861#post-id-287861) two of the most experienced people answering questions (myself and Gijs) have both contributed to and tried to answer. We have both volunteered our time to specifically try to help you answer your question. And we've answered it as best we can based on our undertanding of your question and we've asked for clarifications as to what you consider not answered.

If you want more help on that question you need to take the time and effort to improve your question with clarifications and more details. The better your question the better your answer can be.

> @tfoote Explain to me the upvote of my question and the reason for having multiple answers, then we talk about visibility and incorrectness of my statements.

I do not have to explain anything to you. If you are not willing to abide by our guidelines and follow our etiquette our community we will ask you to not participate. Your expectations of what you can ask for/demand from the community members and trying to [shame](https://answers.ros.org/question/287826/wheres-the-official-python-api-documentation-for-tf/?answer=287873#post-id-287873) people is not productive and is exactly what I'm talking about. Negative and personal attacks are caustic to the community and are not acceptable. They will discourage contributions and poison the whole community. Please consider this an official warning.

nbro, 2018-04-10:
>  If you are not willing to abide by our guidelines and follow our etiquette our community we will ask you to not participate.

Who said I am not willing to at least read your guidelines? Maybe not in the next days, but maybe one day in the future.

It's you the person who thinks I am attacking others. If a person makes fun of me by answering like "interpolation is interpolation", then do not ask me to react like a stone.

> Your expectations of what you can ask for/demand from the community members and trying to shame people is not productive and is exactly what I'm talking about.

My expectations are my expectations, yours are yours. 

> Negative and personal attacks are caustic to the community and are not acceptable.

I have the freedom of speech and I can share my opinion regarding the irresponsibility of people who share with the world non-optimal products. I have the right criticize. 

>  Please consider this an official warning.

This looks like a personal attack. Well, it's a point of view, right? Of course, you're right, because you have "power" in this case. Whatever.