ros/geometry #162: the python listner cannot get the exist transform

AlexWang1996, 2018-04-20:
Hi all, today l find one strange thing when l use tf(python) to get the transform published by my cpp  node. 
[image]
then l use python tf 
[image]
well, some error happens
[image]
However, if l use the cpp node listener , it works.
[image]

;( seems strange and no idea what's wrong 
l am under Ubuntu 16.04, ROS Kinetic

AlexWang1996, 2018-04-20:
[image]
l run rqt_tf_tree to look what happens, and picture tells.

tfoote, 2018-04-20:
Please ask questions on http://answers.ros.org according to our support policy: http://wiki.ros.org/Support We try to keep the issue tracker reserved for bugs and feature developments.

Also mentioned there, please don't use screenshots of text. They can't be searched or reproduced easily. Copy and paste them instead and use code blocks.

Your problem is that you're not giving the buffer any time to populate before querying it. You'll find there are already many similar questions on answers.ros.org with this same problem. Please search there for other releated questions. And see part 3 of this about try blocks: http://wiki.ros.org/tf/Tutorials/Writing%20a%20tf%20listener%20%28C%2B%2B%29 or here: http://wiki.ros.org/tf/Tutorials/Writing%20a%20tf%20listener%20%28Python%29

closed 2018-04-20 by tfoote