ros/geometry #166: Does tf::MessageFilter exist for rospy?

jwhendy, 2018-06-02:
Greetings,

I'm learning ROS with a colleague who's more experienced and as a way to learn quickly, I'm translating some of his code into python. This is also a way to see if there are any deficiencies of python overall for ROS development (he's advocating I learn C++, and I'd hope to stick with a language I already know).

I came across his use of `tf::MessageFilter` and wondered what the official stance is on using it via `rospy`. I saw this on the [C++ api page](http://docs.ros.org/melodic/api/tf/html/c++/):

> The tf::MessageFilter is the recommended method for receiving almost any sensor data from ROS.

I also found [this question](https://answers.ros.org/question/49069/is-there-a-tfmessagefilter-in-the-tf-python-api/) from 2012 suggesting it might not exist for python. It's 6yrs later, so I thought I might just ask at the source.

The quote from the C++ api implies it's pretty important. If it doesn't exist for rospy, is there a desire to have it implemented? Or is the official recommendation to use something like the answer at that question (i.e. roll one's own python implementation)? I would be happy to try contributing, either to code porting or putting a DIY example somewhere on the wiki for reference for other python users.

tfoote, 2018-06-04:
A contribution of a python implementation would be appreciated. In general synchronizing in python is easier and it's ususally used in less pipelineing and latency critical areas so it's been in less demand.

The best place for an implementation to be contributed would be in [tf2_ros](https://github.com/ros/geometry2/tree/melodic-devel/tf2_ros/src/tf2_ros)

Elio-in-summer, 2022-11-27:
> A contribution of a python implementation would be appreciated. In general synchronizing in python is easier and it's ususally used in less pipelineing and latency critical areas so it's been in [...]

Hi, I'm also wondering the same question, so how can I do  synchronizing in python bwteen tf msg and subscriber msg, is there any Doc or tutorial？ 
Thank you

closed 2025-05-31 by sloretz