ros/geometry #176: SIGSEGV in waitForTimeTransform in docker!

mojovski, 2018-09-19:
I am facing a strange SIGSEGV error inside the TF module, when used in a docker container.

**In short:** 
the method ``tf::Transformer::waitForTransform(...)`` causes a segmentation fault!
See this repository to reproduce the error quickly: https://github.com/mojovski/tf_docker_issue

**In long:** 
The ``waitForTransform`` in these lines causes the crash:
```
std::string global_frame="map";
    std::string robot_frame="base_link";
    try{
      ROS_INFO_STREAM("timenow: " << ros::Time::now().toSec());
      bool res=listener.waitForTransform(global_frame, robot_frame,
                               ros::Time::now(), ros::Duration(0.1), ros::Duration(0.1),
                               &err_msg);
    }
    catch (tf::TransformException ex){
      ROS_ERROR("%s",ex.what());
    }
```

Strangely, it runs on my Ubuntu 16.04 machine. But when I also use Ubuntu16.04 inside docker, this crashes.

If you do the steps in the readme here https://github.com/mojovski/tf_docker_issue, you will get this stack trace:
```
Thread 1 "test179" received signal SIGSEGV, Segmentation fault.
0x00007ffff73bddda in ros::TimeBase<ros::Time, ros::Duration>::operator+(ros::Duration const&) const () from /opt/ros/kinetic/lib/librostime.so
(gdb) bt
#0  0x00007ffff73bddda in ros::TimeBase<ros::Time, ros::Duration>::operator+(ros::Duration const&) const () from /opt/ros/kinetic/lib/librostime.so
#1  0x00007ffff65d6013 in tf2_ros::Buffer::canTransform(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, ros::Time const&, ros::Duration, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >*) const () from /opt/ros/kinetic/lib/libtf2_ros.so
#2  0x00007ffff7bb6a3f in tf::Transformer::waitForTransform(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, ros::Time const&, ros::Duration const&, ros::Duration const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >*) const ()
   from /opt/ros/kinetic/lib/libtf.so
#3  0x000000000040a8ac in main (argc=1, argv=0x7fffffffe448) at /root/catkin_ws/src/test179/src/test_bug179.cpp:31
```

mojovski, 2018-09-19:
Closing: If I use the "recent" branch kinetic-devel from here, the error disappears.
https://github.com/ros/roscpp_core

closed 2018-09-19 by mojovski

mfe7, 2018-09-20:
@mojovski I can confirm that I had a similar problem (segfault from tf listener in docker, but it worked fine outside docker), and building roscpp_core from source solved it.

tfoote, 2018-09-21:
There was a unintended ABI change in [the 2018-09-06 kinetic sync](https://discourse.ros.org/t/new-packages-for-kinetic-2018-09-06/5983) involving ROS time. Until the docker images have the old ABI if you install newer packages without doing an upgrade this has been observed to cause segfaults. The docker images usually get regenerated every month or so. The workaround is to just do an apt update && apt upgrade. As also pointed out recompiling from source also fixes it.