ros/geometry #178: A more efficient calculation of quaternion inverse

AndyZe, 2018-11-05:
I'm not an expert on quaternions, but from looking at several online sources, I believe there are 2 equivalent ways to invert a quaternion:

u_x*i+u_y*j+u_z*k - u_w = -u_x*i-u_y*j-u_z*k+u_w

The [tf source code](http://docs.ros.org/jade/api/tf/html/c++/Quaternion_8h_source.html#l00257) uses the left-hand method, which is less efficient (3 numbers are negated instead of 1). If everybody agrees, I could make a PR for this.

Relevant stack exchange proofs (wish I had a better reference):

https://math.stackexchange.com/questions/2016282/negative-quaternion

https://math.stackexchange.com/questions/1937833/the-quaternion-matrix-for-a-rotation-of-theta-2-pi?rq=1

tfoote, 2018-11-08:
This is a fork of the bullet linear math library.

That particular function is here: https://github.com/bulletphysics/bullet3/blob/f7237647fb44f17d791894f5e46be8b80aac3337/src/LinearMath/btQuaternion.h#L497-L505 If that change can be accepted upstream we can backport it here. In general though we've run into trouble diverging from the upstream implementation that I'd rather not encounter again.

davetcoleman, 2019-02-20:
Should we be targeting ``geometry2`` repo for these kinds of improvements? @AndyZe if you're working on moveit1 still using ``melodic-devel``, it will change whether or not its using ``tf1`` or ``tf2`` depending on whether you're compiling on Ubuntu 16.04 (Kinetic) or 18.04 (Melodic)

tfoote, 2019-02-20:
Yes, any changes are highly recommended to be targeted at geometry2 as geometry has been deprecated for quite a long time now.