ros/geometry #189: Drop graphviz dependency from tf

jschleicher, 2019-05-14:
It would be nice to split `view_frames` into a new package `tf_tools` or similar to get rid of the graphviz dependency on headless systems.

(Similar to https://github.com/ros/actionlib/issues/129)

tfoote, 2019-05-14:
Since tf is deprecated for tf2 and in tf2 we've cleaned it up to use tf2_tools for the graphical tools. 

As that cleanup was done as part of the refactor for tf2 I don't plan to refactor this package but recommend migrating to use tf2/tf2_ros which included a significantly reduced dependency footprint in it's design.

closed 2019-05-14 by tfoote

jschleicher, 2019-05-15:
@tfoote Thanks for the hint (I should know, that tf2 is the recommended package): In melodic is  robot_state_publisher and rviz, that still pulls in tf in my setup, so I'll open issues over there.