ros/geometry #192: Potential issue of switched variables

danielwangksu, 2019-07-11:
Not sure if this is a potential issue, the order of variable is reverse to the definition. This could be confusing

```
echoListener.tf.waitForTransform(source_frameid, target_frameid, ros::Time(), ros::Duration(1.0));

bool waitForTransform(const std::string& target_frame, const std::string& source_frame,
                        const ros::Time& time, const ros::Duration& timeout, const ros::Duration& polling_sleep_duration = ros::Duration(0.01),
                        std::string* error_msg = NULL) const;
```
https://github.com/ros/geometry/blob/ea5eb79f479223d2b95f0a5e051554fbaceca7ff/tf/src/tf_echo.cpp#L112

https://github.com/ros/geometry/blob/ea5eb79f479223d2b95f0a5e051554fbaceca7ff/tf/include/tf/tf.h#L206

```
echoListener.tf.lookupTransform(source_frameid, target_frameid, ros::Time(), echo_transform);

void lookupTransform(const std::string& target_frame, const std::string& source_frame,
                       const ros::Time& time, StampedTransform& transform) const;
```
https://github.com/ros/geometry/blob/ea5eb79f479223d2b95f0a5e051554fbaceca7ff/tf/src/tf_echo.cpp#L122

https://github.com/ros/geometry/blob/ea5eb79f479223d2b95f0a5e051554fbaceca7ff/tf/include/tf/tf.h#L242

matt-attack, 2020-07-01:
I've also run into this issue. It has led to a lot of confusion on our end since it returns the inverse of the expected transform.

@tfoote do you know if this is expected behavior for tf_echo or should the order be swapped to match the functions? 

If it is expected behavior that should probably be documented and made more clear.

tfoote, 2020-08-13:
The transform from A to B is the inverse of the transform from B to A. To get the inverse directly you just change the order of the arguments.

If you're printing the transform of Frames A to B it's the inverse of the transform to move data from A to B. Thus when you're echoing the transforms for the frames A to B you have to switch the source and target frames which are specifically referring to moving data from source to target.

The echo command print the transform for the coordinate frames not for the data so they're inverse and thus reversed.

closed 2020-08-13 by tfoote