ros/geometry #201: tf view_frames is broken

Windfisch, 2020-01-10:
Hi,

on ROS melodic running on Arch Linux, `rosrun tf view_frames` does not work for me. I'm getting this output:
```
$ rosrun tf view_frames
Listening to /tf for 5.0 seconds
Done Listening
b'dot - graphviz version 2.42.3 (0)\n'
Traceback (most recent call last):
  File "/opt/ros/melodic/lib/tf/view_frames", line 119, in <module>
    generate(dot_graph)
  File "/opt/ros/melodic/lib/tf/view_frames", line 89, in generate
    m = r.search(vstr)
TypeError: cannot use a string pattern on a bytes-like object
```

By manually patching in the changes from #193, I can resolve this. It would be good to merge that pull request.

xmuller, 2020-06-30:
Up ! 
It would be nice to also merge #193 on ROS noetic. We have got the same error on Ubuntu 20.04.

ana-GT, 2020-07-01:
Agreed, please merge #193. Tested in Ubuntu 20.04 noetic and the patch works (had to change *python* to *python3* but that is specific to my setup).

nilsbore, 2020-09-23:
Another `+1` for merging https://github.com/ros/geometry/pull/193, tested on noetic.

lauralindzey, 2020-10-06:
Big +1 / bump. I also tested #193's fix on noetic, and it worked for me.

I think my setup is pretty bog-standard (Ubuntu 20.04, ros-noetic-desktop-full), and find it surprising that a core ROS debugging tool doesn't work out of the box.

djordjevujic, 2020-10-06:
Another +1 after taking changes from #193  (Ubuntu 20.04, ros-noetic-desktop-full)

Windfisch, 2020-10-06:
@tfoote, can maybe you merge #193 so this issue can be closed? It seems well tested now.

tfoote, 2020-10-07:
#193 isn't dual compatible python 2 and python3 so it couldn't be merged into melodic. But a rebase of it can be merged into noetic now that python3 is required for noetic.

tfoote, 2020-10-07:
> I think my setup is pretty bog-standard (Ubuntu 20.04, ros-noetic-desktop-full), and find it surprising that a core ROS debugging tool doesn't work out of the box.

This tool has been deprecated and it's recommended to use tf2_tools view_frames.py which is why this has received less attention.

lauralindzey, 2020-10-07:
> This tool has been deprecated and it's recommended to use tf2_tools view_frames.py which is why this has received less attention.

Ah, I'd been aware of tf2/tf, but didn't realize that the tf utilities weren't just wrappers around the tf2 equivalents, and that I actually needed to type a few more characters =) 

Do you want a PR that adds a deprecation note to the output of tf's view_frames?

tfoote, 2020-10-07:
That would be great.

JDuchniewicz, 2020-11-23:
The [wiki](https://wiki.ros.org/tf/Debugging%20tools) should be updated to point to tf2_tools package. Otherwise the user could be confused.

ckurtz22, 2020-11-27:
> > I think my setup is pretty bog-standard (Ubuntu 20.04, ros-noetic-desktop-full), and find it surprising that a core ROS debugging tool doesn't work out of the box.
> 
> This tool has been deprecated and it's recommended to use tf2_tools view_frames.py which is why this has received less attention.

I think a reason this fix should be merged is because it is still used in rqt, so the TF frame viewer in rqt is broken.

acarrillo, 2020-12-09:
My team is onboarding from `kinetic` to `noetic` and we are all very accustomed to the old viewer as part of our regular workflow; we are happy to use the new `tf2_tools` provided script, and I think a deprecation warning is a great idea, but leaving it straight out broken will leave a lot of newcomers to `noetic` feeling unsure if the new platform is ready for them yet :/

If we don't want to fix it, I would think you should just outright delete it + replace it with a script that does nothing but print that deprecation warning, but even that is quite severe.

MichaelGrupp, 2021-02-11:
Just saw this issue after opening my PR #222 with a fix, but I agree with @acarrillo that a deprecation warning plus a fix would be a good solution. It's an old habit to use this tool :older_adult:

acarrillo, 2021-02-22:
Indeed, I am still googling for this ticket all the time to re-call what the deprecation alternative is!

Petrox, 2021-04-02:
````
         at line 278 in /tmp/binarydeb/ros-noetic-tf2-0.7.5/src/buffer_core.cpp
Traceback (most recent call last):
  File "/opt/ros/noetic/lib/tf/view_frames", line 119, in <module>
    generate(dot_graph)
  File "/opt/ros/noetic/lib/tf/view_frames", line 89, in generate
    m = r.search(vstr)
TypeError: cannot use a string pattern on a bytes-like object
````

lucbettaieb, 2021-04-23:
Yep
```
luc@p15 ~ $ rosrun tf view_frames 
Listening to /tf for 5.0 seconds
Done Listening
b'dot - graphviz version 2.43.0 (0)\n'
Traceback (most recent call last):
  File "/opt/ros/noetic/lib/tf/view_frames", line 119, in <module>
    generate(dot_graph)
  File "/opt/ros/noetic/lib/tf/view_frames", line 89, in generate
    m = r.search(vstr)
TypeError: cannot use a string pattern on a bytes-like object
```

R I P

lucbettaieb, 2021-04-23:
If we're really trying to deprecate the use of these tf utilities -- is it EOL'd?  When will it stop being released with new versions of ROS1?

EnricoMingo, 2021-08-24:
Hi all, I just came up as well into this issue. Just as info, a simple workaround I did was to add the following line just before ```m = r.search(vstr)```: 
```vstr = vstr.decode('ISO-8859-1')```

JeffXiesk, 2021-11-27:
Hi, I just solved this. 
`sudo gedit /opt/ros/noetic/lib/tf/view_frames

vstr=str(vstr)
`
We add this "vstr=str(vstr)" to make vstr convert to a string and that's ooook!!

tfoote, 2021-12-01:
This was resolved by #222

closed 2021-12-01 by tfoote