ros/geometry #213: tf test_transform_datatypes duplicate name error

flynneva, 2020-06-13:
when compiling from source for ROS noetic, I am seeing the following error:

```
-- +++ processing catkin package: 'tf'
-- ==> add_subdirectory(geometry/tf)
-- Performing Test COMPILER_SUPPORTS_CXX11
-- Performing Test COMPILER_SUPPORTS_CXX11 - Success
-- Using these message generators: gencpp;geneus;genlisp;gennodejs;genpy
-- Found Boost: /usr/lib/x86_64-linux-gnu/cmake/Boost-1.71.0/BoostConfig.cmake (found version "1.71.0") found components: thread system 
-- tf: 1 messages, 1 services
CMake Error at catkin/cmake/test/gtest.cmake:180 (add_executable):
  add_executable cannot create target "test_transform_datatypes" because
  another target with the same name already exists.  The existing target is
  an executable created in source directory
  "/home/flynn/dev/ros/noetic/src/geometry2/tf2".  See documentation for
  policy CMP0002 for more details.
Call Stack (most recent call first):
  catkin/cmake/test/gtest.cmake:89 (_catkin_add_executable_with_google_test)
  catkin/cmake/test/gtest.cmake:37 (_catkin_add_google_test)
  geometry/tf/CMakeLists.txt:86 (catkin_add_gtest)

CMake Error at geometry/tf/CMakeLists.txt:87 (target_link_libraries):
  Attempt to add link library "tf" to target "test_transform_datatypes" which
  is not built in this directory.

  This is allowed only when policy CMP0079 is set to NEW.
```

ROS Noetic (from source)
Ubuntu 20.04
Intel x86

flynneva, 2020-06-13:
resolved compile error by changing geometry2/tf2/CMakelist.txt to:

```
catkin_add_gtest(test_transform2_datatypes test/test_transform_datatypes.cpp)
target_link_libraries(test_transform2_datatypes tf2  ${console_bridge_LIBRARIES})
add_dependencies(test_transform2_datatypes ${catkin_EXPORTED_TARGETS})
```
i'll make another issue over there and close this one.

closed 2020-06-13 by flynneva

flynneva, 2020-06-13:
wow [I forgot I already made a PR to fix this bug](https://github.com/ros/geometry2/pull/465)....long week I guess

maxpolzin, 2021-11-19:
Why did the fix not get merged? I am experiencing the same error on rpi4, raspian(buster), ROS Noetic, TF/TF2 (from source)

adhityab-gda, 2024-10-03:
facts brothher