ros/geometry #227: tf_conversions does not compile under noetic + bionic due python3-pykdl dependency

berndpfrommer, 2021-06-03:
I'm trying to build Noetic on Ubuntu 18.04 because NVidia does not support Ubuntu 20.04 yet on their Jetson/Xavier line.
I am aware that Ubuntu 18.04 is not targeted for Noetic but I see few other alternatives than trying to build it from source on the Xavier AGX platform.

When I follow these instructions: http://wiki.ros.org/noetic/Installation/Source
I hit the following snag:
```
rosdep install --from-paths ./src --ignore-packages-from-source --rosdistro noetic -y
ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
tf_conversions: [python3-pykdl] defined as "not available" for OS version [bionic]
tf2_geometry_msgs: [python3-pykdl] defined as "not available" for OS version [bionic]
gazebo_dev: No definition of [libgazebo11-dev] for OS version [bionic]

```
Under Ubuntu 18.04 there is no such package as ``python3-pykdl``. Any suggestions on how to work around this issue?
Thanks!

saulzar, 2021-08-02:
Same issue.

saikishor, 2021-08-05:
Same issue. Any lead is appreciated!

schmittlema, 2021-08-15:
same issue. Currently the only solution I see available is to build within a docker container.

saikishor, 2021-08-16:
@schmittlema Could you explain us how you are building it inside the container, if the issue is the dependency. We are also trying to build it within the docker, but we are not successful.

Thank you,

schmittlema, 2021-08-16:
Actually since writing this I managed to get it to build without a container. But the container solution would effectively be running 20.04 in a container. There are pre-built containers [here](https://hub.docker.com/_/ros) which seem like the best solution.

In terms of building outside of a container I ran 
`rosdep install --from-paths ./src --ignore-packages-from-source --rosdistro noetic -y -r`
Notice "-r" which continues installing dependencies despite failures. When I went to build the noetic stack, it seems to build pykdl from source bypassing the issue (I think) and builds fine. But I haven't tested running code so I could be giving the wrong advice.

closed 2025-05-31 by sloretz